darthcloud/BlueRetro #345: N64 issue pairing second controller and when pairs it gets unplayable lag

novamax89, 2022-07-31:
I have an n64 with ultrapif and rgb mod. I have 2 n64 switch controllers, I installed it giving 2 different ground pins on esp32, pin 26 and 27 to VIN and 3.3 cables from both controllers to 3v3, and controller 1 to pin 19 controller 2 to pin 5.  It works very good on the first controller but can’t get to work fine on the second one. If the second controller suddenly works fine than the first have massive lag. Never works fine with both controllers. Seems that esp32 is underpowered but it won’t work even if I connect it through usb. 
I tried connecting it in different ways but still nothing fix it. Do you have any fix for this problem?

darthcloud, 2022-07-31:
Yes this is known issue, see #146 

No fix yet

novamax89, 2022-07-31:
Thanks for the reply. I’m sorry I didn’t see it. I subscribed to that and I’m closing this.

closed 2022-07-31 by novamax89