darthcloud/BlueRetro #346: Xbox one controller can't setup trigger L for gamecube

DarkEddo, 2022-08-01:
I can't setup the trigger L on a xbox controller to use it on a gamecube, the other buttons and triggers work perfect. the controller had the last firmware and set up with the Web-Bluetooth (BLE) configuration interface...

darthcloud, 2022-08-01:
can you screenshot your config

DarkEddo, 2022-08-01:
[image]

DarkEddo, 2022-08-01:
i have that setup...

darthcloud, 2022-08-01:
What you are missing is a mapping to map the end of analog stick to the gc digital trigger end.

GC triggers are two buttons in one. an analog part and the digital end. If you use a regular controller and default mapping you only get the analog part.

The end of system manual for GC mention this:
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-System-Specific-User-Manual#mapping-15
```
Using the default mapping, BlueRetro will map the two 
section separately. (Use this if playing with Wii Original 
Classic or TBD Steam controller

Use the "GameCube Merge analog & digital trigger" preset to
 emulate the GameCube trigger using only the analog 
trigger. (Any other type of Bluetooth controller)
```

So in the presets page just load the merge gc trigger preset and you should be fine.

DarkEddo, 2022-08-01:
Thanks, i'm gonna try that...

DarkEddo, 2022-08-02:
I make a test with the indications, in the menu of game (metroid prime 1), the trigger LT work and do the change of screen with no problem... but when i back to the game the trigger don't scanning nothing  even don't select a enemy to shoot...

darthcloud, 2022-08-02:
[image]

If you look again at mapping you got two mapping for LT? One for analog and one for digital?

DarkEddo, 2022-08-02:
yes, i have it... the same with Z button... i need change by digital or analog???

darthcloud, 2022-08-02:
i'll try myself later, if your mapping are exactly like mine that should work as-is with that preset

darthcloud, 2022-08-02:
Work fine for me, tested metroid prime 1 with xbox ctrl and the merge trigger preset, using the scan visor pressing LT to maximum scan items. Can you provide screenshot of the mapping after setting the preset?

Maybe try lower the digital button threshold value from 95% down to 75%.

DarkEddo, 2022-08-02:
Thank you so much!!!... It's working awesome!!!, down the values to 75% and merged trigger prereset y voila!!! works

attach the screen capture

[image]

closed 2022-08-02 by DarkEddo