darthcloud/BlueRetro #347: (Gamecube) Problems with internal blueretro + Qoob Pro Chip

AlucardRetroChallenge, 2022-08-02:
Hi! I'm a gamecube user and installed the Blueretro inside the console (by soldering it to the controller's/memory card's board).
My console has the "Qoob Pro Chip". When I use bluetooth controllers on "Qoob Bios"/"Menu", everything works fine. But... when I use the original Gamecube controller on "Qoob Bios"/"Menu", it gets crazy and pressing buttons randomly (especially the "B" and "A" buttons).
Now I'm obligated to turn the console on with a bluetooth controller, open Swiss GC/Original bios and then use the original wired controller.
I'm using the 1.7.1. version with the "gc_external" file. Is this problem possible to address?
Thanks!

darthcloud, 2022-08-02:
Internal install is not currently supported, if you need original controller you need to either put a switch on ESP power or use the deep sleep function via https://blueretro.io/system.html

closed 2022-08-02 by darthcloud

AlucardRetroChallenge, 2022-08-02:
Do you intend to support gamecube internal instalation in a near Future? I'll try to install a micro Switch as you Said! I'll flash IPL Boot on Qoob Pro to bypass the "Qoob Menu" and avoid the related problems.