darthcloud/BlueRetro #350: 8bitdo N30 Bluetooth unresponsive following reboot

mario-64, 2022-08-12:
Using 8bitdo N30 Bluetooth with Humble's PCE BT receiver (v1.71) on a SuperGrafx. 

Initial pairing worked fine. No issues at all. However, if you power off the console then power it on again the controller is unresponsive. The receiver acts like it connects to the controller (light goes off) yet no input is accepted. It's dead. Power cycling the controller or console doesn't help. The only way to fix this is to freshly pair the controller. Obviously this should not be required every time you power on the console.

darthcloud, 2022-08-12:
Make sure to pair in xinput mode (Start + X). Reconnect doesn't work if you pair in Switch mode with 8bitdo.

mario-64, 2022-08-12:
Thank you. I’ll try that but their documentation for the N30 doesn’t mention xinput mode like some of their other controllers. Maybe that’s just an oversight

darthcloud, 2022-08-12:
If it doesn't might be worth try to update the controller FW, maybe they added the feature later on.

mario-64, 2022-08-12:
Did that too. I'm running v6.16. Actually the N30 doesn't even have an X button. It's quite similar to an OEM 2 button PCE controller. It has A & B with "turbo" buttons right above them. That's why I wanted to use it. The only difference is it also has  L/R shoulder buttons as well.  https://www.8bitdo.com/N30-NS/

mario-64, 2022-08-12:
Closing this as I found the following statement on 8bitdo's website. It seems to be designed only for Switch Online.

Q) Can I connect this controller to Android, Windows 10, macOS, Raspberry Pi, iPhone, iPad, PS3, PS4, XBOX one, XBOX 360, Wii, Wii U directly?
A) No, you can not.

closed 2022-08-12 by mario-64

darthcloud, 2022-08-12:
For now you will need to always pair if using it with BlueRetro.

I will try to figure why reconnect fail in Switch mode later.

darthcloud, 2022-10-11:
For now you will need to always pair.

I will try to figure why reconnect fail in Switch mode later.