darthcloud/BlueRetro #351: PC Engine GamepadAlt mode dpad doesn't work

mario-64, 2022-08-12:
Testing a Retro-Bit Genesis bluetooth controller with the Humble PCE BT Adapter. Both on latest firmware. The controller works correctly when Mode = Gamepad (2 button). However, if you select Mode = GamepadAlt (6 button) the dpad doesn't work at all, even though dpad mapping is identical to Gamepad mode. I *think* button mapping works but can't really test since the dpad doesn't work.

darthcloud, 2022-08-12:
Which game you used?, 6 buttons mode only work in specific games.

mario-64, 2022-08-12:
Couldn’t get into a game. This is just on the SHDS3 Pro menu. Can’t navigate without dpad.

darthcloud, 2022-08-12:
You need to set 2 btns mode while in your SHDS3 menu, then once game started, switch to 6 buttons via web, save and hit reset on the adapter.

The problem is SHDS3 do not support 6 buttons mode in it's menu.

mario-64, 2022-08-12:
Really? Wow I didn’t know that. Will just stick with 2 button then. Thank you

mario-64, 2022-08-13:
I checked on the TO Discord and the SHDS3 Pro has no issues in the menu with 6 button controllers. 

"'ve had no problem with a wired Hori fighting commander. It's my main pce controller"
"i also have used my hori 6 button just fine in the menu"

darthcloud, 2022-08-13:
Is the controller set in 2 btns mode while they do this or in 6 buttons mode?

mario-64, 2022-08-13:
Doomy Dormer will be reaching out to you. He reproduced the issue using a DS5. He said it looks like the 6 button controller mapping is all messed up. Could you please re-open this? Thank you

DoomyDoomer, 2022-08-13:
I can confirm 6 Button mode doesn't work correctly in either SHDS3 PRO or Ever Drive menus but 2 button mode works fine. I also tested Street Fighter II on both and both have the same mapping issues. The options menu shows a 6 button controller but mapping isn't correct.

darthcloud, 2022-08-16:
Anyone with this issue can ping @DoomyDoomer directly for an hardware fix.

Closing here since not a software bug.

closed 2022-08-16 by darthcloud