darthcloud/BlueRetro #354: N64 controller port 1 not working after flashing v1.63 up to the latest

axs8336, 2022-08-14:
I have tried the flashing v1.6.3 up to the latest v1.7.2 and controller port 1 does not work. The other three ports work fine. I flashed an earlier version and the controller port 1 works but when using a wired controller the wireless controller doesn't work properly

darthcloud, 2022-08-14:
Yes this is due to the #264 workaround I'm planning to remove it in v1.8 if all goes well.

Internal install are not yet supported and as such its normal you get problem when using wired controller.

axs8336, 2022-08-14:
Oh ok. Thanks for replying. I’ll be waiting for the update. Have great weekend.

axs8336, 2022-12-14:
Thanks a lot for let me know. I will definitely give a try. Keep up the great work.

Closed #354 as completed.