darthcloud/BlueRetro #355: always gets esp32 reboots

Radeon91, 2022-08-15:
hi.
Tried to build this one but i always gets the esp32 rebooting, check voltage, straigt pins, different firmwware version but the results are always the same.

https://github.com/Micha213/BlueRetro-PS1-2-Receiver

[image]

What can i do?

darthcloud, 2022-08-15:
Look like power issue like https://github.com/darthcloud/BlueRetro/discussions/170

maybe @Micha213 can help?

Micha213, 2022-08-15:
Yes, that sounds like a power problem. Did you connect 8v and GND (both) correctly? Can you please take a picture of the installation?

Radeon91, 2022-08-15:
Tried with 8 volts from ps2 and also 8v and 3.3v from a power supply,
always same result.