darthcloud/BlueRetro #365: The psx/ps2 can't use analog's L2 R2

ekang4887, 2022-08-19:
I use PS4 or XBOX one pad to connect the blueretro ,use psx mode. When play gt4/gt3,I found I can't useL2 R2's analog, I set the R2 as the gas throttle, and it wii go the max,not linear.
Is it normal or if it can be setup to analog button? thank you

darthcloud, 2022-08-19:
PS2 Pressure buttons is not yet supported.

Alternatively you can set the game tu use right stick for throttle and break.

And then using the button mapping function map right stick up on R2 and right stick down on L2.

Note that doing this, pressing both simultaneously the mapping lower in the list win.

ekang4887, 2022-08-29:
ok,I will try it.thank you

DJm00n, 2023-02-03:
@darthcloud is there any chance that this will be implemented eventually?

darkblood159, 2023-02-06:
Yea i was also wondering if there will be a fix soon for this issue? i just ordered one of these blueretro devices and im a bit nervous on how this will effect my gameplay and how often.

YgorSan, 2023-07-16:
I agree, from what I understand there are some games that are incompatible with controllers that do not support this type of feedback (perhaps an unsuccessful strategy by sony and game developers at the time to prevent the advancement of sales of low quality generic gamepads). Call of Duty Big Red One and Call of Duty 3 are some of the games that currently don't work with blueretro, maybe it's because of that function so at the moment it's a bit sad to see this kind of incompatibility

ButtonBasher92, 2023-12-30:
Just checking in to see if there has been any progress on this? it's the only thing stopping me from using my Retroscaler Blueretro on my ps2 over the Pademu feature on OPL that does support it, Though Pademu does break some games. Blueretro would be the better option. That and proper rumble support.

the ability to use the PS3 controllers pressure sensitive face buttons like the original ps2 controller would be great.

darthcloud, 2024-01-03:
This beta add support for ps3 ctrl pressure button while using a PS1/PS2 system:
https://github.com/darthcloud/BlueRetro/actions/runs/7398955532

Everything can be remap at will as usual and the remap will be analog if the source and destination are analog.
Turbo work as well on pressure buttons.

If you use a modern controller without pressure button, you can map a joystick direction or one analog trigger to one of the PS2 pressure button and it will be analog.

ButtonBasher92, 2024-01-03:
I am busy testing the Beta as I type this message, Using a DS3 controller on my PS2 and all the pressure sensitive face buttons work as they should. seems the Rumble function has been improved also? it doesn't feel so violet anymore.

Ordering another Retroscaler Blueretro mini now for Player 2. 
Great improvement! thank you!

DJm00n, 2024-01-05:
@darthcloud It finally works! I tried to test in [PadTest](https://www.psx-place.com/resources/ps2-controller-tester-by-jbit.670/) PS2 tool and now I can see button values on DualShock4: 0 or 255 for usual buttons and a full 0..255 range on L2/R2.

[image]

Its a long-waited feature. Thank you!

PS: Proper vibration support on PS2 is still missing. According to the PadTest readme:

>To test vibration, hold 'triangle', then move the left joystick down for the strong motor (that seems to have 96 positions), or press up and down on the directional pad for the weak motor (that has 2 positions).

Both, `strong` and `weak` motors are working but with same strength. I think this is a last major feature that may be improved for PSX/PS2.

DJm00n, 2024-01-05:
There is list of games that are supporting pressure sensitive buttons:
https://www.pcgamingwiki.com/wiki/List_of_games_that_support_Pressure-Sensitive_Face_Buttons#PlayStation_Emulation
https://www.pcgamingwiki.com/wiki/List_of_games_that_support_Pressure-Sensitive_Face_Buttons#PlayStation_2_Emulation

DJm00n, 2024-01-23:
@darthcloud looks like this could be closed now?

darthcloud, 2024-01-23:
@DJm00n don't worry about this, I close bugs once an official release is out.

darthcloud, 2024-02-19:
[Fixed in v1.9.2](https://github.com/darthcloud/BlueRetro/releases/tag/v1.9.2)

closed 2024-02-19 by darthcloud