darthcloud/BlueRetro #372: BlueRetro directly wired for P1, P2, and P3 but often fails to pair if P2 and P3 are unplugged

rs1n, 2022-08-25:
I cut 3 controller extension cables and wired them directly to the ESP32 using jumper connectors. When all three connectors are plugged into P1 through P3, everything appears to work as expected. However, if P1 (which was wired to supply 5V) is plugged in but P2 and P3 are unplugged, then controllers will only very rarely pair. And when they do, there is no response from pressing buttons or using joysticks (I have been using a PS4 controller).

I flashed with the Dreamcast-specific firmware (non-universal). My ESP32 is from Mouser (via the link on the BlueRetro Github page). My wiring is as follows:

P1 5v - J2 pin 19
P1 D0 - J3 pin 6
P1 D1 - J3 pin 2
P1 GND/SHIELD/SENSE - J3 pin 1

P2 D0 - J3 pin 5
P2 D1 - J3 pin 10

P3 D0 - J3 pin 9
P3 D1 - J3 pin 2

Again, if I disconnect D0 and D1 for P2 and P3, then plugging in only P1 works without issue. But if D0 and D1 for P2 and P3 from the extension cables are connected to the ESP32 all while P2 and P3 NOT plugged into the console, then pairing often fails. And when it succeeds, the behavior is as described above -- no input is recognized, and eventually the controller disconnects (PS4 blue light is off). Everything works as usual with P1 P2 and P3 all plugged into the console (in all combinations of P1 through P3 being paired or left unpaired). 

Even though the wires for P1 through P3 are connected to the ESP32, I would expect the behavior to be the same as if only P1 was wired to the ESP32 when leaving P2 and P3 not plugged into the console.

darthcloud, 2022-08-25:
Unfortunately this isn't supported, BlueRetro got no way of knowing if a port is plug or not.

As such unused port pin need to be tied to ESP32 3.3V.

So if a port is populated it needs to be connected.

A workaround this would be to use a DTDP switch on the extra port to toggle the pins between the plug and 3.3V

rs1n, 2022-08-25:
> So if a port is populated it needs to be connected.

I see now. I'll implement a switch as you suggested. Thank you for all your work -- BlueRetro is such an amazing project!

darthcloud, 2022-08-25:
I'll probably come up with a new version of the external HW that borrow bits of the internal HW design to add detection of 2P-4P plug.

darthcloud, 2022-12-14:
v1.8 release now support port detection which allow to build an external adapter which plug are not required to be connected.

https://github.com/darthcloud/BlueRetro/wiki#2---building-hardware-hw2

https://github.com/darthcloud/BlueRetro/releases/tag/v1.8

closed 2022-12-14 by darthcloud