darthcloud/BlueRetro #381: snes mouse disconnects

DigiDwrf, 2022-08-28:
I'm using a Logi M585 bluetooth mouse with my snes and it works perfectly, but passed a white, it stops working, so I have to reset my mouse pairing mode and starts working again. I'm using the last firmware (1.7.3). Other devices work OK, but 8bitdo controllers might fail sometimes.

darthcloud, 2022-08-29:
If you can, disconnect the 5V from snes and plug the USB to a PC and get the logs:
https://github.com/darthcloud/BlueRetro/wiki/Getting-BlueRetro-debug-logs-via-Serial-port-Windows-10

Press reset (EN) on the ESP32 so I can see the boot logs, then connect the Mouse and then wait until the mouse disconnect. Save full log to a txt files and upload here by drag & drop.

DigiDwrf, 2022-08-29:
awesome, I will. Do I have to keep the blueretro device connected into the snes while I do all of this?

darthcloud, 2022-08-29:
You don't if you can reproduce without it.

DigiDwrf, 2022-08-29:
OK, what about the 3.3v that feed the 74AHCT125N controllers? I think they still have to be connected, right?

darthcloud, 2022-08-29:
leave everything as-is just don't plug it into the snes.

DigiDwrf, 2022-08-29:
[LOG.txt](https://github.com/darthcloud/BlueRetro/files/9447083/LOG.txt)

darthcloud, 2022-08-29:
Thanks, I think the mouse doesn't like when pairing mode ends. I'll look later what I can do.

DigiDwrf, 2022-08-29:
@darthcloud thank you

DigiDwrf, 2023-07-17:
any luck on this issue? with other bt mouses are supported?

darthcloud, 2023-10-07:
Sorry for not getting back to you on this.

Try this v1.9 beta:
https://github.com/darthcloud/BlueRetro/actions/runs/6437347093

I moved the end of pairing mode to be earlier.

DigiDwrf, 2023-10-07:
thanks! I'm gonna check it out 👍🏻

DigiDwrf, 2023-10-10:
I just tested it but it's the same, after 30 seconds, mouse stops responding, turning mouse off and on makes it work again, and 30 seconds later stops, and so. Any other bluetooth mouse I could use? I tried using a cheap chinese bt mouse from AliExpress, it never pairs 😂 but it works fine with windows.

darthcloud, 2023-10-10:
I use Logitech MX Master 2S

darthcloud, 2023-12-02:
So I bought this mouse M585 and it work flawless in v1.9!

Can you try v1.9 with it?

I used it non stop 5 min no issue. Then left it idle 1 min, and it worked right away when i touch it again.

If it still doesn't work for you maybe check with the Logi tool on Windows if you can somehow update the mouse?

I got another Logi mouse, the M720 that behave exactly like you said, I'll try to update that one later and see.

DigiDwrf, 2024-01-16:
Awesome! Yes, I tried it and it kept stopping again, then I used my pc as a middleman bluetooth mouse with a tool called across center, and it worked great. https://www.acrosscenter.com/

haxman, 2024-06-15:
Any updates on this Darthcloud. I recently got the M720 and it does the same thing as DigiDwrf and you have mentioned.
Thanks,

tomorrowis2014, 2024-08-03:
I'm also having this problem with an onn. mouse I got from walmart. I've tried a Steelseries Aerox 9 (pairs, doesn't seem to provide input), the Across Center software (pairs, only keyboard registers input and seems to be stuck drifting to the left), and this "onn. 5-Button Wireless Bluetooth Ambidextrous Mouse, USB-A Nano Receiver, Black" mouse. I'm using the Bitfunx 4-port N64 adapter from Amazon, it has its own bootleg FW bin, but the official HW1 can be flashed to it and pair at least one controller. This is a livable compromise as I would like to play 64DD software with a mouse more than I really need to play N64 with an Xbox controller. I can upload a Putty log for it if it seems like something that can be fixed, or a compatibility list would be awesome as I'm really not trying to spend a ton of money on a mouse I'm not otherwise going to use a lot (hence not buying an actual N64 mouse). Any assistance is much appreciated.

Thank you.

darthcloud, 2025-01-06:
I think I figured that one. My M720 now work.

Can you guys try this beta?
https://github.com/darthcloud/BlueRetro/actions/runs/12636840722/artifacts/2391429328

(You need to login to github for the link to work)

DigiDwrf, 2025-01-06:
Sure, I will!

darthcloud, 2025-01-24:
Fixed in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)

closed 2025-01-24 by darthcloud