darthcloud/BlueRetro #391: Stock esp32 chips not working?

scp-electrocution, 2022-09-01:
I bought eighteen chips and everything has always worked perfectly, lately I have taken another six (all the same) I can program them successfully but I cannot connect with the controllers and I cannot enter the settings menu.
Any ideas to solve this problem?

log flash tool:
C:\blueretro\BlueRetro_psx_external.bin
1 kill
2 kill
3 kill
test offset :  4096 0x1000
case ok
test offset :  32768 0x8000
case ok
test offset :  65536 0x10000
case ok
test offset :  4096 0x1000
case ok
test offset :  32768 0x8000
case ok
test offset :  65536 0x10000
case ok
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
Compressed 23824 bytes to 15084...
Compressed 3072 bytes to 153...
Compressed 539248 bytes to 328611...

 is stub and send flash finish

https://www.amazon.it/gp/product/B08CCYWZN3/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

darthcloud, 2022-09-01:
make sure you use 1.7.3, 1.7.2 had a issue with newly flash chip.

scp-electrocution, 2022-09-01:
I use the latest rev 1.7.3 but I have also tried the older versions, the problem remains

scp-electrocution, 2022-09-01:
I try to order some official chips, thanks

closed 2022-09-01 by darthcloud