darthcloud/BlueRetro #394: external wii works on snes mini, but not on wii with wiimote.

davidxgames, 2022-09-05:
i am trying external blueretro wii, and it works fine on snes mini, but on wii it doesn't work, i am using ps4 ps5 and xbox one controllers, but it doesn't work. I am using an esp32 dev kit.

darthcloud, 2022-09-06:
Yes currently only the 8bit data mode is supported. So only NES and SNES mini are currently supported.

While I might add support for the regular bit mashup protocol for sake of completes, I don't thing using BlueRetro with a Wiimote is an overall good idea.

davidxgames, 2022-09-07:
thanks for replying, so the best option for wii is to use blueretro in gamecube mode.

darthcloud, 2022-09-07:
I'll keep this open as it is a valid issue even if the use case is suboptimal.

GC mode is best option for Wii.

darthcloud, 2022-09-29:
[v1.8-rc2-4-g2f6c943_hw1_wii_extension.zip](https://github.com/darthcloud/BlueRetro/files/9674820/v1.8-rc2-4-g2f6c943_hw1_wii_extension.zip)

Can you try this? This build should work on wiimote

darthcloud, 2023-01-15:
v1.8.2 now support the 6 bytes mode the Wii use https://github.com/darthcloud/BlueRetro/releases/tag/v1.8.2

BUT it's not supporting encryption, so Wii menu and most commercial games will not work.

Only homebrew will work.

#545 will track the feature to add encryption

closed 2023-01-15 by darthcloud