darthcloud/BlueRetro #398: Saturn LS adapter seems to cause short/reset when connected.

MooThing-Code, 2022-09-10:
Based on the soldering diagram on the wiki, I created a pair of Sega Saturn adapters for the project. with all the level shifter ICs installed, the appropriate connections to 3v3 and ground made and all the wires soldered to their points the cable will immediately cause a reset on the console and sinks a lot of current through the cable (enough to perceptively feel heat radiating from it). 

This continues to happen as long as 5V and ground are wired up to the board, the rest of the data pins seem to be uninvolved in this issue.

Is the current diagram for the LS board appropriate for the Saturn, or have I just miswired it twice in a row?

darthcloud, 2022-09-10:
Trace pitch is quite small, make sure your PCB manufacturers didn't short some trace!

darthcloud, 2022-09-10:
[image]

MooThing-Code, 2022-09-10:
I'm hoping the PCBs aren't the issue here, I've made both a Mega Drive and Super Nintendo adapter using the same boards and ICs.

I'm reading a resistance of about 470kΩ between the 5v and GND after thoroughly cleaning the board, along with all of the non grounded IO lines and GND.

MooThing-Code, 2022-09-10:
[image]

darthcloud, 2022-09-10:
Connect only gnd and 5v to start with no BlueRetro (cable only), if that work add each cable one by one until you find issue . Add BlueRetro at the end.

MooThing-Code, 2022-09-10:
My issue is that even with only gnd and 5v connected this issue still occurs.

darthcloud, 2022-09-10:
Probably a short under one of the hip. Use a blank pcb forst and connect GND and 5V only to validate empty pcb is fine.

Then two options:
either remove chip one by one testing between each on previous board. 

Or build new one adding chip one by one.

MooThing-Code, 2022-09-10:
I gave it a shot with a bare PCB and a DB25 connector, 5V was making its way to the esp32 (I'm using the blueretro AIO mainboard) but the AM1117 regulator on the PCB was blisteringly hot. It didn't reset the console however, so it's different.

MooThing-Code, 2022-09-10:
Put another one together, the ICs literally went up in smoke. I give up.

darthcloud, 2022-09-10:
Sorry really no idea, maybe try pcb ls v1.1 as it is the board I used for saturn. Or use the AIO one.

closed 2023-04-01 by darthcloud