darthcloud/BlueRetro #400: GameCube controllers stuck

Moadib6, 2022-09-14:
When I'm playing GameCube at some point, the controller stays on and stuck, like on standby. but the controller does not react and I have to turn off the console to make it work again.

I use an xbox one controller and a ps4 controller, with both controllers this error occurs.

Moadib6, 2022-09-23:
- I turn on my gamecube
- sync PS4/XBOX-ONE controller
- start a gamecube game
- at the moment when I'm playing the game, the PS4/XBOX-ONE controller gets stuck, it's not always like that
- I have noticed that it happens when you have to press many buttons at the same time for long periods
- the control does not turn off it just stops responding and does not let you do anything else
- I solve it, I turn off the gamecube, turn it back on and sync the controller and everything works fine

darthcloud, 2022-09-23:
No my question is how you installed your blueretro. Take picture of the esp32 where I can see all connection easily.

Moadib6, 2022-09-24:
These are the pictures of the installation
[image]
[image]
[image]
this is the diagram i made
[image]

darthcloud, 2022-09-24:
Leaving the ESP32 in static generating foam is rally bad.

This is not an official espressif board and people often got issue with those.

Try getting an official one from digikey or mouser.
ESP32-DEVKITC-32E with ESP-WROOM-32 (Get it on [Digikey](https://www.digikey.com/en/products/detail/espressif-systems/ESP32-DEVKITC-32E/12091810?s=N4IgTCBcDaIKYGcAOBmMBaAJnAbgawEsAXAY3TThAF0BfIA) or [Mouser](https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-32E?qs=sGAEpiMZZMv0NwlthflBi3XYxq%252BTapwyHnR2%2FrAf4bY%3D))

Moadib6, 2022-09-24:
Thanks for the help, I'll remove the sponge and get an official esp32

krikris, 2022-09-24:
Source of the diagram: https://www.mundoyakara.com/2022/02/controles-bluetooth-en-nintendo.html

Moadib6, 2022-09-26:
ok, i will follow this diagram

closed 2022-12-14 by darthcloud