darthcloud/BlueRetro #405: ESP32  ESP32-D0WD-V3 (revision v3.0) doesnt work - flash successfull

Ntemis, 2022-09-21:
esptool.py -p /dev/ttyACM0 \
  -b 460800 --before default_reset --after hard_reset --chip esp32 \
  write_flash --flash_mode dio --flash_size detect --flash_freq 40m \
  0x1000 bootloader/bootloader.bin \
  0x8000 partition_table/partition-table.bin \
  0x10000 BlueRetro_universal.bin
esptool.py v4.3
Serial port /dev/ttyACM0
Connecting.....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:e6:86:05:c6:10
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00094fff...
Compressed 23824 bytes to 15084...
Wrote 23824 bytes (15084 compressed) at 0x00001000 in 0.8 seconds (effective 253.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 153...
Wrote 3072 bytes (153 compressed) at 0x00008000 in 0.1 seconds (effective 339.3 kbit/s)...
Hash of data verified.
Compressed 544320 bytes to 332238...
Wrote 544320 bytes (332238 compressed) at 0x00010000 in 7.8 seconds (effective 556.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Ntemis, 2022-09-21:
lsusb outputs
# Bus 003 Device 029: ID 1a86:55d4 QinHeng Electronics USB Single Serial
dmesg
[ 4751.815531] usb 3-3.1: new full-speed USB device number 28 using xhci_hcd
[ 4751.917165] usb 3-3.1: New USB device found, idVendor=1a86, idProduct=55d4, bcdDevice= 4.43
[ 4751.917179] usb 3-3.1: New USB device strings: Mfr=0, Product=2, SerialNumber=3
[ 4751.917185] usb 3-3.1: Product: USB Single Serial
[ 4751.917189] usb 3-3.1: SerialNumber: 5185031746
[ 4751.926843] cdc_acm 3-3.1:1.0: ttyACM0: USB ACM device

Ntemis, 2022-09-21:
I have no led pulsing and when i try to do anything in https://blueretro.io
i get this in logs
Web Bluetooth API is not available. Please make sure the "Experimental Web Platform features" flag is enabled.

darthcloud, 2022-09-21:
This just means your browser dont support web Bluetooth, use chrome.

You actually wired a LED on IO17 ?

Ntemis, 2022-09-21:
No the board has a red led that is always ON , i thought that one should been the pulsing one

Ntemis, 2022-09-21:
Actually i did tried to pair my ps4 and it pairs relatively fast but  doesnt show up in js-test
https://jstest-gtk.gitlab.io/

darthcloud, 2022-09-21:
not sure what you doing but BlueRetro is for retro console, not PC