darthcloud/BlueRetro #410: Turbografx 16

warcraft78, 2022-09-23:
Some games make adapter freak out and drop controller,  it won’t sink up to anything until after you get past the troubled area..  might and magic 3 at player setup, power drift at player setup are a few examples.  I’m using latest firmware

darthcloud, 2022-09-25:
Issue reproduceble, will investigate later this fall.

warcraft78, 2023-03-22:
Any chance on seeing the issue?

MeatyMuffin, 2024-05-28:
Has any progress been made with this issue? I’ve seen it myself with several different games on SuperGrafx hardware via a Turbo Everdrive Pro in my testing in the past few days.

The most recent games I’ve found the issue with are the two Sherlock Holmes: Consulting Detective games in which the issue will present itself upon attempting to load any spoken dialogue. Once the issue hits the dialogue will be skipped entirely while the adapter will reset itself and drop the paired controller. Once repairing the controller the game will work as intended until another selection is made then the whole string of issues will repeat.

One final thing I’ve noticed that might be unrelated to the issue itself is that while the majority of CD based games will not let you access an in game menu for the Everdrive via button combo all the games I’ve experienced this issue with will and through a full version of the menu as if the disc image is being read as card based media.