darthcloud/BlueRetro #416: Nes mini controller to Wiimote on N64

Invictaz, 2022-09-29:
I connected a Wii Classic Pro (OEM) controller to the Wiimote. That works fine on the N64 via BR.

Then I disconnect the Wii Classic Pro controller and connect a fake Nes mini controller to the Wiimote. It does nothing. 

The Wii Classic Pro controller has much more buttons and even analog sticks. 

Is this because it's a fake controller or isn't it supported yet?

darthcloud, 2022-09-29:
I guess they didn't bother implement the regular 6 bytes mode.

This build support the 8 bytes mode, but only if I can confirm it on after it's enable. Which they might have skip.

Anyway try this build:
[v1.8-rc2-4-g2f6c943_hw1_n64.zip](https://github.com/darthcloud/BlueRetro/files/9674924/v1.8-rc2-4-g2f6c943_hw1_n64.zip)

Invictaz, 2022-09-29:
I cannot try the build as my RetroRosetta only has bluetooth OTA for the moment. Flashing the separate partition table is hard without a CP2102. The one I have might output 5V says GrechTech which can fry the controller ports he said.

Can you upload a firmware that is flashable via OTA?

darthcloud, 2022-09-29:
all firmware are updatable via OTA, just use the bin file.

the other files are only for initial usb flash for a blank esp32.

Invictaz, 2022-09-29:
@darthcloud with your firmware the fake NES mini controller plugged into the Wiimote started working. However, I have very weird stick drift in Perfect Dark. The character walks around in circles. And the button mapping is weird.

I tried doing live output over bluetooth but that doesnt seem to work on the Retrorosetta. Also not on the older 1.7.3 firmware.

darthcloud, 2022-09-29:
I think it required the 8 bytes mode config to start working but then probably report itself to be still in 6 bytes mode which would explain what you see, ie a 8bytes frame decoded as 6 bytes.

I'll send you another build to test later.

darthcloud, 2022-09-29:
try this
[v1.8-rc2-5-gca5d95a_hw1_n64.zip](https://github.com/darthcloud/BlueRetro/files/9677843/v1.8-rc2-5-gca5d95a_hw1_n64.zip)

Invictaz, 2022-09-29:
Thanks for the fast update. Some good and bad news: 

Found out that the fake NES mini controller I have mostly looks like the 3rd party Orzly branded controller.

The stick drift after your update is still there. However, it can be mitigated partially: reconnect the classic pro controller again to the Wiimote and press sync on the WiiMote and then swap back the Orzly nes controller = drift stops. But not in Perfect Dark. I have only tested this succesfully in Mario 64. Perfect Dark still drifts. 

So if this firmware did change anything I'm not sure.

Invictaz, 2022-10-20:
Tested with Wii RVL-005 OEM Nintendo controller which is mostly the same as the official SNES mini controller. No issues. Need to test further.

darthcloud, 2023-01-15:
v1.8.1 now support this controller https://github.com/darthcloud/BlueRetro/releases/tag/v1.8.2

Note that you will need to use a custom mapping to map the a & b button correctly if you use NES

closed 2023-01-15 by darthcloud