darthcloud/BlueRetro #425: 8bitdo M30 D-pad will not respond in Turbografx

Violator-80, 2022-10-07:
The 8bitdo M30 works, but only the buttons will respond. I have attempted to remap the d-pad, but that has had no effect. I have successfully used PS4 and Retro Bit Saturn controllers with my Turbografx, but the M30 d-pad continues to be a challenge. 

Both the M30 and Blue Retro are on the latest firmware. I am using a SSDS3 in my setup.

vicsidious, 2022-10-12:
Hey Violator, 
stumbled upon your post and I MIGHT have a solution for you! I had the same issue with one of my 8bitdo controller and it happened to be because the M30 firmware allows you to map the dpad either to the left joystick, the right joystick or the dpad itself. In my case, the dpad was mapped to the right stick.
 
You want the m30 dpad mapped to the dpad, and not to its analog stick default! 

Here's how you do it (courtesy of the 8bitdo website!)

"**The default mode of the DPad on this controller is as left analogue stick**. You can also:
LEFT + Select : set Dpad as left analogue stick.
Up + Select : reset Dpad.
Right + Select : set Dpad as right analogue stick.
Down + Select : Swap A/B and X/Y mapping (on Switch mode only).
*Press and hold any of the key combinations above for 5 seconds to map the buttons.
*LED will blink in red to indicate the success of each button mapping.
*You need to reset the buttons manually."

Hope it helped :)
-David

Violator-80, 2022-10-12:
Thank you for your reply, it was a big help. I did a Left + Select combo, got the red LED confirmation and then did a Up + Select combo, got the red LED and it started working. I had never seen the information you provided so I am very grateful that you shared it. Take care.

vicsidious, 2022-10-12:
With great pleasure! You can have more informations here: https://support.8bitdo.com/faq/m30-bluetooth-controller.html
Not working for 8bitdo, but I enjoy their products a lot :D
Have a good one!

darthcloud, 2022-12-14:
Thanks @vicsidious!

closed 2022-12-14 by darthcloud