darthcloud/BlueRetro #427: Non Issue:  Add PS2 Remote Power On/Off

coolduder1001, 2022-10-07:
I'm sorry for putting this here, I don't know how to change the code to add this.  It would be great to add a remote on off to the PS2, the reason being that no additional hardware should be needed, other than some wires.  This would obviously only work if the esp32 is internally installed.

You would need to read the voltage of either the green or red LED to detect when power is going to them, and a connection to the power button.  If the red LED is on && green LED is off, hold button for 2 seconds.  If the red LED is off && green LED is on, hold button for 5-6 seconds.  Hold PS or Xbox button on your controller for 3 seconds to activate the on off function.

luizoti, 2022-10-07:
I suposse you can rename 'Non Issue' to 'Feature Request'.

darthcloud, 2022-12-14:
Internal install is now supported in final v1.8 release
https://github.com/darthcloud/BlueRetro/releases/tag/v1.8

However PS2 was not tested and no instructions will be provided beside the HW2 spec https://github.com/darthcloud/BlueRetro/wiki#2---building-hardware-hw2

Only attempt if you know what you are doing!