darthcloud/BlueRetro #428: Add Sixaxispairtool for Android to pairing guide on wiki

Invictaz, 2022-10-09:
The pairing guide here 

https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#ps3

describes Windows as the only method to pair a Dualshock 3. In order to update the master bdaddr. This is incorrect. It can also be done on any phone or tablet running Android, no root needed, only a USB OTG cable and a USB-A to mini usb male cable.

Just press pair in the app and input the bluetooth address from the webUI from blueretro.io

The tool is available from the Internet Archive (no virusses found):

https://archive.org/details/SixaxisControllerV1.1.3

This step skips the need for a Windows computer. 

Another option is Sixaxispairer for Linux:

https://github.com/user-none/sixaxispairer

darthcloud, 2022-10-09:
Nothing say its the only way.

It's simply one of the way (with OSX) that I documented.

closed 2022-10-09 by darthcloud

Invictaz, 2022-10-09:
I  wonder why you sound offended. I just offeren another way.