darthcloud/BlueRetro #437: DC being detected as JVS

sonik-br, 2022-10-20:
Hi!

I've bough an assembled dreamcast cable (dual DC > DB25) and it's not working.

**When debugging I've seen that it's detecting as JVS.**

This is a common problem? Anything I can do to fix?

I only have another cable, for the saturn. And this one is working perfectly fine.

```
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_INQUIRY_COMPLETE
GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
GPIO[22]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
GPIO[25]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
GPIO[32]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
GPIO[34]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
# Detected system : 14: JVS
GPIO[25]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
# BT_HCI_EVT_LE_ADVERTISING_REPORT
BAD
BAD
BAD
BAD
BAD
BAD
BAD
```

darthcloud, 2022-10-20:
You likely inverted the port 1 two data line

sonik-br, 2022-10-20:
Bought it assembled... It's possible to test with a multimeter without disassembling it?

sonik-br, 2022-10-21:
@darthcloud have you ever seen this?

https://user-images.githubusercontent.com/16923826/197304618-e7db3b20-994b-4d79-854d-7ad5ac7b52ec.mp4

A slight touch on the db25 plastic shell makes it reset (and detect as JVS).
I'm not applying any pressure. Just a very soft touch.

darthcloud, 2022-10-21:
It's clearly defective, and I do not provide support for those scammer using the project name as their company name. Ask questions to to people you gave money to.

Sorry

closed 2022-10-21 by darthcloud

sonik-br, 2022-10-22:
Oh man, I though it was from one of the chinese sellers that sponsors the project. But clearly it's not. :-(