darthcloud/BlueRetro #438: 8bitdo M30 and SF30 d-pad up direction always pressed

biotenshu, 2022-10-21:
Flashed 1.3.7 firmware and tested on Dreamcast, PS1 and PS2. DualShock 4 works great. With 8bitdo M30 and SF30, d-pad up direction registering as pressed when it isn't. Very similar to this issue: https://github.com/godotengine/godot/issues/44721.

darthcloud, 2022-10-21:
Did you pair it in Xinput mode? I got the sf30 and it should work in Xinput mode.

biotenshu, 2022-10-21:
> Did you pair it in Xinput mode? I got the sf30 and it should work in Xinput mode.

Yes. X + Start. Recorded:
https://www.youtube.com/watch?v=2kpEshhRn2o

biotenshu, 2022-10-21:
Same with SF30. Japanese Dreamcast VA1.
https://www.youtube.com/watch?v=cDZyaNnHtzU

LastLifeLeft, 2022-12-10:
Got the same result with both my M30

darthcloud, 2022-12-12:
Internet say the M30 as various dpad mode, you need to switch it to true dpad mode:
SELECT + DPAD UP for 5 seconds

I don't have the controller so please refer to 8bitdo instruction.