darthcloud/BlueRetro #441: Inconsistent reconnection with previously paired Xbox One controller

jesus-barrera, 2022-10-22:
Hello, I'm having trouble using the RetroScaler Converter for PS1 and PS2 (I'm using it on a PSOne) with an Xbox One Controller. I can pair the controller correctly and use it with no problem (for a while, at least, more detail later), but when I try to reconnect it after powering off and on the console, it will not always reconnect: sometimes it does, sometimes it doesn't. I have to reboot the BlueRetro or unplug and plug it again, sometimes multiple times, before it reconnects. It appears to be related specifically to the Xbox Controller since it doesn't happen with an 8Bitdo M30 controller I have. I have the latest firmware on both, the BlueRetro (psx_internal v1.7.3) and the Xbox One Controller (v5.13.3146.0). 

These are the logs for a successful connection:

[successful connection.txt](https://github.com/darthcloud/BlueRetro/files/9843399/successful.connection.txt)

And these are for a failed connection:

[failed connection.txt](https://github.com/darthcloud/BlueRetro/files/9843401/failed.connection.txt)

I actually got these powering the BlueRetro via micro usb without plugging it to my PSOne, but it exhibits the same behavior (I don't know if it is safe to plug the BlueRetro to the PSOne and connect it to the PC at the same time, please let me know if it is). To reproduce this, I simply power the BlueRetro (by turning the console on, or connect it to the PC in this case) and then turn on the Xbox One controller.

Also, after playing for a while, maybe 30 minutes or so, the controller stops working, the game (Colin McRae Rally) says "Please replace your controller" as if it was disconnected, but the BlueRetro and the Xbox One Controller are still paired. This doesn't happen with the 8Bitdo controller neither. I've seen similar problems in other issues but none of them mentions the reconnection problem nor provide the logs, so I opened this one.

Awesome project, thanks for your time.

psicotron3, 2023-01-11:
My blueretro is a gamecube external firmware h1 v1.8.1 (using it on a wii) and this is happening for me to with a xbox one S controller. Started after updating controller to version v5.13.3146.0. 

For me, the reconnection problem isn't as bad (don't appear very often), but my controller stops after just a few button presses in the same way described.

I have another Xbox One S controller that's still on firmware 4.8.1932 and it is working "fine" (connects and stays connected), but using this one, blueretro reset when Xbox button is pressed. The controller with firmware v5.13.3146.0 doesn't have the reset issue. 

Overall, firmware 4.8.1932 works better on other devices (e.g. firmware v5.13.3146.0 breaks compatibility with 8bitdo adapters for switch). If it was not for the reset issue, that would be the case with blueretro too.

dleer, 2023-09-13:
I have a BlueRetro for N64 and see this as well. I have both a DualShock4 and a Xbox One controller on the latest firmware. Both these controllers pair initially without problem but the XBO controller typically reconnects only after several attempts. Sometimes power cycling the BlueRetro gets it done quicker.

konwektor, 2023-12-10:
II got simillar issue in gamecube , and Dreamcast, with xbox controllers. If I use combo "Disconnect all controllers (LT+RT+START+A)  esp32 is disconnecting all controllers, and waits for BT connections from already paired gamepads.(because still got power).
After that power off console or unplug BR from socket.
 This way if i switch on gamepad, it recconnects quite fast to blueretro. 
On Other way use boot btn on esp32 to disconnect paired controllers, next time if reconnecting takes more then 10s, reeboot esp32 (push EN button) wait max10sec, and voila - reconnectred.
This Issue is only on xbox gamepads, no mather what system, no difference is external or hw2.

Best solutions in my case - use combo to disconnect

darthcloud, 2024-10-06:
I feel this got better in recent firmware, can you retry in v24.10?

konwektor, 2024-10-15:
For me, xbox one s gamepad now is reconnecting without problems on BlueRetro 24.04 (24.10 didnt check).
Problem is coming when before connecting to BlueRetro already paired xbox gamepad, i got ps4  or nintendo wiiu gamepad connected, those are constantly sending update status and this is blocking xbox gamepad to reconnect or even pair.

closed 2025-01-24 by darthcloud