darthcloud/BlueRetro #445: *RUMBLE* Reconnection needed on PS2 using DS3 when rumble is activated in web interface

vicsidious, 2022-10-26:
After turning on the Rumble feature in the web configuration interface: 
When I turn the PS2 on, then sync a previously paired Dualshock 3 controller, it syncs correctly (first led on the DS3 controller turns on and the blue light on the ESP32 stop flashing and turns off) but no input is registered.

I have to reset the ESP32, and drop all existing connections (left button, and right button twice on the ESP32) and sync the controller again for it to work. 

Simple enough workaround, but it might indicate an issue somewhere that needs to be corrected, thought I'd bring it up :)

vicsidious, 2022-10-27:
EDIT: Test have been done with DS4 and DS5, with identical behavior. Tests also shows that simply unconnecting and connecting the ESP32 back on the wired adaptor solves the issue, in a similar fashion as a ESP32 reset using the buttons.

darthcloud, 2024-05-29:
I completely rewrote the rumble support, so maybe this beta can help

https://github.com/darthcloud/BlueRetro/actions/runs/9286878322

darthcloud, 2024-10-06:
Fixed in v24.10:
https://github.com/darthcloud/BlueRetro/releases/tag/v24.10

closed 2024-10-06 by darthcloud