darthcloud/BlueRetro #456: (PS1) Analog button doesn't work to switch between d-pad and left joystick

vcesne, 2022-11-04:
Hello,

Analog button doesn't work to switch between d-pad and left joystick.
Tested with PS4 and Xbox serie controller.

darthcloud, 2022-12-14:
Can you provide more details? It work for me.

Which game?
Which buttons you press on PS4 and xbox controller?

vcesne, 2022-12-14:
It happened with Tony Hawk 2 and Ridge Racer Type 4.

I pressed Xbox button and PS button to switch between d-pad and left joystick without success.

I have the same problem with a Psone and a PS1.

Invictaz, 2023-01-15:
Interesting find, need to hear more about this.

darthcloud, 2023-01-15:
@Invictaz dont spam bug tracker with useless comments

darthcloud, 2024-04-20:
The Xbox Home button used to be a bit bogus in older version but in recent BlueRetro FW like v1.9.2 used in conjunction to latest Xbox controller firmware from the [Xbox accessories Win10 app](https://apps.microsoft.com/store/detail/xbox-accessories/9NBLGGH30XJ3) I think that should be fine now.

If not feel free to reopen the issue.

closed 2024-04-20 by darthcloud