darthcloud/BlueRetro #459: ps4 joystick not connecting

jairslira, 2022-11-08:
my ps4 joystick does not connect, it works normally both on the ps4 console and on the pc, but not on the blueretro, I have already reset the joystick and the blueretro has the latest firmware version, I want to know if someone can help me, thank you in advance for your attention

cermeza, 2023-03-19:
In blue retro you have six axis tool for ps3 joystic work for ps4 but if you long press pair buton wait until blinking then pres share and pslogo on controler and hold until  light bar start flashing fast it will conect fast I haved same problem with manual on blue retro
Regards

darthcloud, 2024-04-20:
@jairslira what model of PS4 controller you got? Original from Sony? What model number?

jairslira, 2024-04-21:
Hello, I saw that the problem was actually with the control itself, which
was defective, thanks for your attention

darthcloud, 2024-04-21:
No worries, thanks for the update

closed 2024-04-21 by darthcloud