darthcloud/BlueRetro #461: (Fully) Support for 8BitDo SN30 Pro2 Controller

NeoRame, 2022-11-12:
the Controller seems to work nearly fine BUT in "N64 Mode" i cant switch the Rumble/MPak modes withe the Home/8BitDo Button.

when i try to change the controller starts to reconnect and the game freezes (sometimes Crash!) till the contoller ist back. but in this case it has trouble to find the rumble contoller. the MPak controller is workink but the virual MPak seems to be faulty for the most games.

darthcloud, 2023-01-15:
Can you try v1.8.2? https://github.com/darthcloud/BlueRetro/releases/tag/v1.8.2

If it doesn't work get me a bluetooth trace on Win10 for each mode supported by the controller:
https://github.com/darthcloud/BlueRetro/wiki/Bluetooth-HCI-trace-with-Win10

darthcloud, 2023-10-06:
No trace provided, feel free to reopen with one.

closed 2023-10-06 by darthcloud