darthcloud/BlueRetro #473: Switch Pro Controller Stops

Neurotic8883, 2022-11-25:
After a certain amount of playing on my PS1, the game stops responding to my controller, but connection is maintained between my Switch Pro controller and the BlueRetro Dongle. If I hit the reset button on my controller, the dongle and the controller disconnect and resync, but the game remains unresponsive, as if the controller was unplugged.

Hitting the reset button the BlueRetro dongle and waiting for the controller + dongle to pair up allows me to play again. Sometimes I get a couple of hours of gameplay before this happens, sometimes 20 minutes. I haven't noticed it on my Dual Shock 4 but I need to test more.

DorDur215, 2022-12-16:
This issue has been fixed for me with the 1.8 firmware update that's been released a few days ago. At least with my aliexpress gamecube dongle and switch online snes controller.

wtfbbqhax, 2022-12-16:
1.8 still has this problem

qwertymodo, 2023-02-18:
My Switch Pro controller is constantly disconnecting from my GC internal adapter. Most times it will reconnect after a few seconds, but sometimes it requires a full power cycle of the console to get it to reconnect. Running the latest firmware (1.8.3).

darthcloud, 2023-12-09:
Fix is now in v1.9.1

closed 2023-12-09 by darthcloud