darthcloud/BlueRetro #483: Dualsense simulated click for Gamecube

coolbho3k, 2022-12-12:
I've seen this requested a few times. It would be great to have an option on the Dualsense to simulate the tactile bump at the end of the L/R button travel for Gamecube controllers.

I want to look into this myself - the only hardware I have is the Laser Bear internal PCB which is coming today, as long as I am able to figure out how to compile for/flash it I'm good to do some experimentation.

coolbho3k, 2022-12-12:
`bt_hidp_ps5_set_conf` is I think analogous to the `outputReport` data structure in DS4Windows: https://github.com/Ryochan7/DS4Windows/blob/master/DS4Windows/DS4Library/InputDevices/DualSenseDevice.cs

So we must write to a bunch of the bytes in `tbd1` to set trigger effects.

coolbho3k, 2022-12-21:
I got my ESP32 dev boards in earlier this week, and I'll look into it over the holiday. Thanks!!!

Invictaz, 2023-01-15:
@coolbho3k any updates?

coolbho3k, 2023-01-15:
I didn’t have time to play around this over holidays, but planning on
getting to this soon!

coolbho3k, 2023-03-12:
Sorry for late update. I finally got time to work on this and got this working. I'll have a PR later this week.

coolbho3k, 2023-03-12:
See https://github.com/darthcloud/BlueRetro/pull/640 for a preliminary version of this. I'd like feedback on how we'd configure this. 

I haven't tested on my actual GameCube yet, but I can feel the trigger click with my bare ESP32.

Invictaz, 2023-03-12:
Isnt it so that in the GameCube games the effect only works on the left or right trigger?

coolbho3k, 2023-03-13:
@Invictaz My intention was to simulate the digital "click" at the bottom of the travel on original GameCube controllers. It's on both the L and R trigger on the GameCube.

Invictaz, 2023-03-13:
Yeah but I know of some controversy around the Super Mario Sunshine game that the click was needed and some emulators or controllers did not operate it well enough.

coolbho3k, 2023-03-13:
OK, I think the PR is mostly working now.

closed 2023-03-19 by darthcloud