darthcloud/BlueRetro #494: Tons of visual analog noise after connecting BlueRetro in GameCube

Mitch200584, 2022-12-14:
It used to be crystal clear, but now there is a layer of subtle static all over the screen. Most noticeable on GameBoy Interface HF when using a Flashcart on the blue homescreen. I use EZ Omega Definitive Edition.

darthcloud, 2022-12-14:
Sound like possibly a power related issue.

Which type of GameCube you got? DOL-001 or DOL-101?

Are you using an official Nintendo power supply? (DOL-002)

darthcloud, 2022-12-14:
Just to confirm, I assume RGB scart cable means you got a PAL cube with RGB on the analog port? (So no video mods)

Mitch200584, 2022-12-14:
I have a NTSC GameCube. It supports RGB if it has the serial port

Mitch200584, 2022-12-14:
I forgot to mention the GCHD MKII. Thats how I plug my Scart cable into it

Mitch200584, 2022-12-14:
Words From Mike Chi (RetroTink5X):  
"The EON device is inherently noisy. My guess is it’s picking up whatever extra noise is being generated. Obviously in an ideal world the blue retro won’t be generating noise, & if it did, the EON would be designed properly to filter it. But these things happen."

So I don't know this firmware fixable or maybe a future revision could tackle this, but yeah, unfortunate for EON/RGB Scart users. I really did like the Console reset feature :P

darthcloud, 2022-12-15:
For sure I can do much about that software wise, might be best to contact laser bear directly as they designed the hardware.

Extrems, 2022-12-20:
I couldn't measure a difference before and after installation of the GameCube BlueRetro Internal Adapter using the FFT function of my Siglent SDS1104X-E (200 MHz unlocked), with the EON GCHD Mk-II Rev 1, HD Retrovision Wii Component Cable, RGBench, and a Wii U Pro Controller connected. The normal spikes at 54 MHz and above can be observed, but nothing new and even less so within nominal video bandwidth.

I couldn't notice a difference on the RetroTINK-5X Pro either.

Extrems, 2022-12-20:
I've done more measurements using the Bitfunx GameCube Component Cable, Retro-Bit Retro Prism Component Cable, and the Nintendo GameCube Component Video Cable. I'm now pretty confident this is not a design flaw of the GameCube BlueRetro Internal Adapter.

darthcloud, 2022-12-29:
Thanks for the test result.

Likely something specific to that console.

Can't do much myself about this so closing.

closed 2022-12-29 by darthcloud