darthcloud/BlueRetro #496: NES with 8bitdo DIY NES controller

Hattmeiny, 2022-12-14:
I have installed the ESP32 Devkit in my AV Famicom with it flashed to the NES 1.8 HW1 Firmware.  The controller I'm using is an original AV Famicom dogbone controller that has been swapped to the 8bitdo DIY Bluetooth PCB.  Everything connects as it should except I get no in put from the "B" button on the controller.  If I go into the web IO and remap B to A and A to B, it does just that. Then A button doesnt do anything.
I tried this same controller on a different Blueretro device (PS1) and both buttons do work with that system.
I tried a PS5 controller on my AV Famicom and it uses both A and B buttons correctly.

Has anyone used one of these DIY kits before with your NES/Famicom?  Anyone experienced this and know how to correct?

I have used the 8bitdo DIY on my Genesis 6 button and it works fine.  Took forever to figure out the button mapping, but all buttons work correctly.

vicsidious, 2022-12-14:
Hi! I had the same issue, with the same diy kit from 8bitdo, and it was a real head scratcher!

Why doesn’t it work?: long story short, the diy nes pcb  “B” button seems incorrectly mapped on the “Y” button (IIRC), which is not used by the NES implementation of BlueRetro.

What can I do to solve the issue?: well, after a LONG time of fiddling with the firmware code, I noticed the issue and remapped it INSIDE the firmware and recompiled a new version of it for myself. But now that the firmware 1.8 allows per-system configuration, you could technically remap the “B” button to the correct input and save the config FOR THE NES only, to your Esp32.

If you need more help, let me know! 
-David

vicsidious, 2022-12-14:
Quick note: I didn’t create an issue about the matter because I considered the anomaly to come from the 8bitdo DIY board and not from the BlueRetro firmware.

darthcloud, 2022-12-15:
If one of you can provide a full BlueRetro log while that controller connect I can chande the default in next release.

https://github.com/darthcloud/BlueRetro/wiki/Getting-BlueRetro-debug-logs-via-Serial-port-Windows-10

Hattmeiny, 2022-12-15:
Hey Vicsidious,
Thanks for replying and giving me hope!  
I'm not entirely sure how to remap to make this B button into Y.
When I go to the mapping under Advance Config on the webio, I can choose Source and Destination.  Since I'm using NES to NES, that is what I clicked.  The resulting buttons to map between are only A and B.
I'm not sure how to get to any additional settings to remap the Y

Hattmeiny, 2022-12-15:
I was able to get it working by first using a dreamcast as the source to NES and changing the dreamcast button for NES B until it worked.  I used the mapping sheet to find out what that should be and manually changed it.  Dreamcast A is NES B and Dreamcast B is NES A

darthcloud, 2023-01-15:
v1.8.2 now support this controller natively. https://github.com/darthcloud/BlueRetro/releases/tag/v1.8.2

The controller need to be configured to expose the d-pad as a d-pad:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#73---d-pad-as-joystick-or-d-pad-configuration

closed 2023-01-15 by darthcloud

Hattmeiny, 2023-01-15:
Good monring.  I updated my NES blueretro to the above 1.8.2 and do not see an option under the SRC or DST for the 8bitdo DIY kit.  When I choose NES and NES under SRC and DST, it doesnt work correctly.  What am I missing?

darthcloud, 2023-01-15:
The point of the update is that the default mapping work.

Just load the default gamepad preset, no manual mapping needed.

darthcloud, 2023-01-15:
If it doesn't work after you load default preset let me know

Hattmeiny, 2023-01-15:
Hey, it appears to work fine.  B and A both function.  One thing of note, I have an NESRGB installed.  The controller would typically allow for palette switching with a combo of Start Select and Left/Right to change the pallete.  I appear to have lost this once changing to blueretro.  I tried with multiple controllers.  This of course might not be a Blueretro issue, but it appears to be directly related.  Any ideas?  Thanks for all of your help

darthcloud, 2023-01-15:
BlueRetro is slower than TTL chip for its signal edge transition so if that device read exactly at the clk edge the data the wouldn't be ready yet, it takes like 300ns - 500ns.

pkerga, 2025-08-28:
Hello
I'm using the latest firmware 25.04 and I still seem to have this same problem with my 8bitdo kit for dogbone. I was also able to fix the problem by using the Dreamcast B = Nes B setting. It's strange because it works fine with the NES controller kit.