darthcloud/BlueRetro #497: Blueretro Gamecube Internal: Xbox Controller. Pressing the Xbox button resets blueretro and current game

andrew171717, 2022-12-14:
I'm not sure if this is intentional or an issue, but pressing the xbox logo button causing the game to restart and the blueretro to disconnect all controllers and reset. I could not find any documentation on this. Attempting to remap the xbox button with blueretro.io does not seem to change anything.

andrew171717, 2022-12-14:
Also, pressing X button does not seem to be registering as anything. Pressing the buttons on a controller tester shows some strange/broken button mapping (https://www.gc-forever.com/forums/viewtopic.php?t=2305). I was able to remap Xbox:B to GC:B, but pressing Xbox:X shows up as nothing and attempting to remap it with blueretro.io results in still no button press on the GC. I have "Xbox One S | S/X" selected as Src and Gamecube selected as Dest.

darthcloud, 2022-12-15:
Is your controller updated to latest FW, BlueRetro only support latest.

See https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#xbox-one-s-series-xs--adaptive-controller

andrew171717, 2022-12-15:
Potentially not, I'll check and report back. Thanks

closed 2022-12-15 by andrew171717