darthcloud/BlueRetro #499: Retrofighters Brawler NSO Controller - Pairs OK, inputs not recognized

Squishlord26, 2022-12-17:
Thanks for your work on this project. I've been using an 8bitdo ultimate controller the past few days and wow. Playing original hardware from 15 feet away without cords is amazing.

I just received my new controllers today in the mail. I am able to pair the brawler NSO controller to the console, but none of the inputs are being recognized.

I've tried switching the controller between its two layouts, with no success.

I've also tested the controller on the switch, it does function correctly while paired to the switch. Could I be missing something?

Thanks in advance.

darthcloud, 2022-12-21:
I ordered one, so will probably be fixed in a month or so.

vicsidious, 2022-12-31:
Is that the entirely white one? If so, I ordered it a few weeks ago and it sits in Canada Post warehouse somewhere in Montreal, 3 hours from where I am! So, chances are, I'll get it in a few days! I'll back you up on my findings!

joeyfpe, 2023-01-03:
i have the same issue

skite2001, 2023-01-04:
@Squishlord26 
same for me, got the grey one in combination with an aliexpress receiver. pairs without problems but no button control.
@darthcloud
awesome, can't wait for your fix ^_-

vicsidious, 2023-01-06:
I FINALLY recieved mine. It took long enough... So, I too can confirm the issue with the controller pairing but not registering any inputs. The issue seems to be the same with the "Duelist", another bluetooth gamepad from Retrofighters, but with a Gamecube layout. (like this: https://retrofighters.com/our-collection/duelist-super-smash-bros-pc-switch-controller/)
So, I can't solve the issue, unfortunately, but I can certainly offer my help to test the new firmware with the Duelist, since I suspect that the issue is the same with both controllers.

joeyfpe, 2023-01-15:
right about an hour ago firmware v 1.8.2 hw1 went live.
my setup now works with sm64.
I got the firmware through a link in the device web config
flashed the N64.bin firmware
then went to presets and selected global config
and mapping config output1 -> N64 ->RetroFighters Brawler64 Xinput -> save
inputs were then working apart from camera and punch which was resolved buy switching the controller mode from pro to n64. hope this helps :) happy gaming

darthcloud, 2023-01-15:
The controller is supported in two ways in v1.8.2

Switch mode & Xinput mode:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#111---first-pairing

Switch mode support rumble. and can use the default mapping

In Xinput Rumble dont work and you need to use a special preset: RetroFighters Brawlers64 Xinput

Xinput rumble issue tracked in #546

closed 2023-01-15 by darthcloud