darthcloud/BlueRetro #510: A better way to mapping buttons?

hardlevel, 2022-12-22:
It is possible to make a way to map buttons in an easy way?

What I have in mind:
Some way to select the console that the adapter will be used, ps1, ps2, n64, etc

It tells the order for mapping buttons, we take note of it somewhere and enable a mapping mode on the next turn on or when it detects the controller

Connect the Bluetooth controller and start mapping the buttons within the order provided earlier

I think it can take the process of mapping more "universal" and easier

Any chances?

Invictaz, 2023-01-15:
Mapping like the Brook adapters (not sure if patented) is super good.

Just press button combination to arm remap mode. Press the buttons to swap or combine into one. And then save (optional) and then exit remap mode with a button or button combination. Super easy. Maybe change the leds on the Blueretro to show in Mapping mode or not.

hardlevel, 2023-01-16:
i think its not patened, but a better system for mapping can make every controller compatible

WolfRamiO74, 2024-01-03:
mine adapter remap is not even working, when i save the remap setting on the web, its say "changes saved" but when you open it again every change is like the default.

darthcloud, 2024-04-20:
GameID feature since v1.8 let you create per game profile for n64, ps1 and GC. For other system it minimally let create a profile for that console if you are using the universal adapter.

Some improvement will come in the future but I don't when.

closed 2024-04-20 by darthcloud