darthcloud/BlueRetro #511: BlueRetro 1.8.1 N64- Accessory type "Both" breaks Rumble and Memory pack functionality

wtfbbqhax, 2022-12-22:
After upgrading the firmware to 1.8.1 I started seeing games report corrupt memory pack. I continued testing and found Rumble was also not working.

I've narrowed this down to the Accessory type "Both" and have attached a video as well. 

This was working correctly using the previous firmware. 

https://user-images.githubusercontent.com/7323455/209241560-5619aa19-da60-45d2-bee1-7dee72dc2944.MOV

darthcloud, 2022-12-22:
Dont use both, n64 is not a dreamcast it got only one slot, set either none, rumble or mem.

Docs define valid value for N64
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-System-Specific-User-Manual#accessories-14

closed 2022-12-22 by darthcloud

wtfbbqhax, 2022-12-22:
Ack, thanks for that