darthcloud/BlueRetro #512: Feature Request: Button Mapping Combo for UltraHDMI and N64D OSD

Sowden, 2022-12-23:
Hey there guys.   So I just got my blueretro working externally on my UltraHDMI N64.  I saw in the web config interface that the Mapping Config page can config some mappings for the N64.  Could I request a feature for a one press combo that brings up the UltraHDMI OSD?  The button combo is L/Z + Dpad Right + C Right + R Trigger.  The N64 NSO controller has a ZR button on the top that currently does the same mapping of Z.  That would be a perfect button to use to bring up the UHDMI OSD with the greatest of ease.  It would be helpful if there was a button mapping combo for the N64D as well.  I don't own one and I don't know what the button combo is, but I'm sure someone here could chime in and inform us of that.  But yeah, just a thought if it's possible.  Thanks for such a great product.

Sowden, 2023-01-17:
Hey there.  So I have been looking at button combos a little more closely today.  What I would like to have is have one button press three or four buttons at once to have something happen.  For my UltraHDMI I would like the (N64 NSO controller) ZR button to bring up the OSD.  But I am now looking at buying a one controller dongle for my Gamecube, so that I can connect my 8bitdo DIY replacement board (both my NES and Genesis controllers) to my Gamecube and use it with my emulators.  However I am running into the reverse problem where I would like to press three buttons and it pressing the c-right button on the Gamecube controller.  If you look at raphnets NES to Gamecube adapter, it has this feature.  I thought that I would be able to achieve this by having you create a preset.  But all the presets seem to do is just manipulate the "Src" and "Dest" in the Advance Config menu.  So am I right that at this time there is no way to create any multiple button presets to just one button press, and in vice versa?

darthcloud, 2023-01-17:
clicking the + button add a fresh mapping line leave everything default exept src and dsc

You src will always be for example ZR in all
then you put the destination to one button of the combo

Example:
SRC DST
ZL L
ZL R
ZL Start
ZL A

with that mapping pressing Z will result in combo L+R+Start+A

Sowden, 2023-01-17:
Hot damn son.  It worked.  That might be added to the manual as well.  Thanks for your brilliant product dude.

closed 2023-01-17 by darthcloud

Sowden, 2023-01-17:
Alright, so I started thinking again, and thats never good.  But I was thinking about the vice versa on my Gamecube.  I want the normal setup

SRC DST
D-pad right D-pad right
Start Start
A A
B B

But I also would like a combo to work
D-pad right C-right
Start C-right
A C-right

So in this senerio if I input all of this, when I hit A, will it hit A + C-right, or will it be looking for the combination of all three buttons to press C-right?

darthcloud, 2023-01-17:
You can't define a macro to press a button.

Pressing A in your example will just press A and C-right

Sowden, 2023-01-17:
Right on.  Are there any plans to implement that in the future or is it just not within our grasp?