darthcloud/BlueRetro #522: liblfds is no more in GitHub and .gitmodules should be modified

JPZV, 2022-12-28:
[liblfds](https://github.com/liblfds) and its repository [liblfds7.1.1](https://github.com/liblfds/liblfds7.1.1) are no more in GitHub as they moved it to https://liblfds.org/git/liblfds7.1.1, so the .gitmodules should be modified to point to the new URL.

But, this may cause an issue on Windows Machines as that repository includes tests and benchmarks with (ridiculously) longs names and path, causing the error of `error: unable to create file liblfds7.1.1/test_and_benchmark/libbenchmark/src/libbenchmark_benchmarks_queue_umm_enqueue1_then_dequeue1/libbenchmark_benchmarks_queue_umm_pthread_spinlock_process_private_enqueue1_dequeue1.c: Filename too long`.

This issue means that cloning this repository (BlueRetro) may need some workaround.

darthcloud, 2022-12-28:
Thanks for the head up, I will just switch to my own github fork of it.

JPZV, 2022-12-28:
Thanks!

Not sure if it's related, but when I try to compile this repository using a clone for the new liblfds7.1.1, I get this error:

```
d:/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: BlueRetro.elf section `.iram0.text' will not fit in region `iram0_0_seg'    
d:/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
d:/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 188 bytes
```

188 bytes is so tight so maybe there're some new code/features added to liblfds7.1.1

darthcloud, 2022-12-28:
Make sure you build using the same esp-idf version I got.

See here for dev setup docs:
https://github.com/darthcloud/BlueRetroRoot

Latest esp-idf added some bloat somewhere and IRAM doesn't fit since last week.
I will fix that when I update to latest in January.

Otherwise I pushed a new ref for the repo to use my own fork fixing this issue.

Thanks!

closed 2022-12-28 by darthcloud

JPZV, 2022-12-28:
> Make sure you build using the same esp-idf version I got.
> 
> See here for dev setup docs: https://github.com/darthcloud/BlueRetroRoot

Yeah, noted that, I'll try to compile with that as I'm using the ESP-IDF extension for Visual Studio Code.

About the original issue, thanks for fixing it!!! I'll try to pull it right now!