darthcloud/BlueRetro #523: Random disconnects using GameCube blueretro and GBros adapter

hugo-brito, 2022-12-29:
I'd recently got a Gamecube Blueretro board from laser bear industries: https://www.laserbear.net/products/gamecube-blue-retro-internal-adapter
I updated it to version 1.8.1.

Tried it out with a couple of controllers, namely the switch pro controller, the Xbox elite 2 controller and two GBros adapters (most recent firmware v2.26).

The Xbox controller seems to be the most reliable, easy to setup and satisfying to play.
The switch pro controller, unfortunately misses inputs, and shows noticeable lag.
But what really bothers me are the GBros adapters. This is because it seems to me that using them is the only way to have a wireless yet, most authentic experience when playing the GameCube.

They often don't connect when I turn on the console (need to pair them all the time, or retry several times), and when I'm playing they randomly disconnect. The only way to get them back to working state is to get up from the couch and do a couple of power cycles on the console because, most likely, it won't just work at the first attempt.

I've been using them on the X input, which seems to me that requires least tweaking.

Since all the controllers I mention here show none of the issues mentioned here in other consoles/computer, I can only assume the issue is with the board.

Looking forward to hearing out about a solution. Cheers.

TheSloshedPanda, 2023-01-10:
I am having similar issues with my pro controllers. It seems that anything 'made for' Nintendo switch has horrible reliability issues. I can pair my two Xbox one controllers just fine, or ONE switch pro controller, or ONE powerA GameCube controller.

My powerA or switch controllers work fine, but only in single player, and if I try to pair a switch pro controller with any other controller, or another pro controller, it drops out after a few minutes.

Hopefully someone can help, because this is worthless to me as of right now.

hugo-brito, 2023-01-12:
How's the lag when you use the switch pro controller? I experience very noticeable lag and key misses.

I feel the same as you, this GameCube board underdelivers...

CapitaineBidouille, 2023-01-14:
1 switch pro alone seems reliable.
2 at the same time starts to add noticeable lag but not at the point that many disconnection occurs
3 at the same time = very unstable, 1 controller will disconnect after a few seconds...

Using laserbear internal blueretro kit tested on MKDouble Dash with Swiss rev1400

CapitaineBidouille, 2023-01-14:
Just got an answer from LaserBear support.
The bug on Switch Pro controllers is known. They are working on a fix, look at this issue : https://github.com/darthcloud/BlueRetro/issues/146

The support agent told me "You can make multiple switch pro controllers work, but will take multiple connections and disconnects to get things going. "

TheSloshedPanda, 2023-01-14:
Laserbear emailed me back, confirming

"There seems to be a bug in the release firmware (up to 1.8.1) that causes switch controllers to disconnect when two or more are connected. Darth Cloud is currently investigating the issue and hopes to get it resolved for the next firmware update. I am sorry about this, previously it was only 3rd party switch controllers that had this issue which is why I did not list those as strictly compatible. A single switch controller mixed with other's from sony or microsoft will work."

I bought from their site, and have a 30-day return window. I guess I'm going to return it. That really sucks though, I had such high hopes, and I know that @darthcloud is trying to resolve it, so no blame there.

CapitaineBidouille, 2023-01-14:
At least the blueretro kit does not replace or alter the behaviour of my original controllers. I'm a bit disappointed that something that look so trivial does not work yet out of the box just like you. I will keep mine and wait for an update. Until then, for multiplayer I will use the GC wired controllers.

Playing on GameBoy Interface with the Switch Pro controller is really amazing. Before that, I was using a real GBA (which is great but it uses the battery of the console while playing and requires time each time to go on the menu of the flashcard to boot the GC controller stuff) or the GC controller which is far from ideal. The switch pro seems to be the perfect controller for that retro games, the D-Pad of it is very pleasant to use and the controller is comfortable on my hands.

hugo-brito, 2023-01-14:
I'm happy that issue is getting attention.

I'm hoping that both the multi-Switch Pro/lag issue gets fixed, and that a similar fix is also applied to the GBros.

I'll be writing to laser bear to make them aware of the issues on the GBros as well.
Looking forward.

CapitaineBidouille, 2023-01-14:
> I'm happy that this issue is getting attention.
> 
> I'm hoping that both the multi-Switch Pro/lag issue gets fixed, and that a similar fix is also applied to the GBros.
> 
> I'll be writing to laser bear to make them aware of the issues on the GBros as well. Looking forward.

That issue is major in my opinion since the GC BlueRetro adapter is marketed to work just fine on Switch Pro official controllers : https://www.laserbear.net/products/gamecube-blue-retro-internal-adapter , which is not in the facts really true today (only in single player, I tried to connect them multiple times as advised by the support agent, without getting success so I will continue my wired controllers in this case).

We do not have many choice, I analyzed the controllers market : 
- 8bitdo controllers tends to have similar issues on multiplayer
- PS5 dualsense controllers is the best in terms of latency but it is really expensive and the battery life is pretty bad for a bluetooth controller (6 hours)
- XBox controllers seemed to be a great alternative, but I preferred to go on getting Switch pros so I can also use them for multiplayer with friends on my switch without having to buy extra adapters 
- Third party controllers are overall often crappy comparated to their official counterpacts

darthcloud, 2025-01-24:
Do you still have this issue with [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01) ?

Can you try both Xinput & Switch mode.

TheSloshedPanda, 2025-01-24:
> Do you still have this issue with [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01) ?
> 
> Can you try both Xinput & Switch mode.

Awesome, I'll have to check tomorrow. If this has been fixed that would be AMAZING.

I want nothing more than to have 4 PowerA GameCube controllers paired at once.

darthcloud, 2025-01-24:
@TheSloshedPanda This issue is about Gbros adapter.
4 PowerA won't work due to #146

darthcloud, 2025-04-16:
Feel free reopen if v25.01 didn't fix this

closed 2025-04-16 by darthcloud