darthcloud/BlueRetro #526: Mapping for a N64 Controller

JPZV, 2023-01-02:
I'm making a re-map for a [Bluetooth N64 controller](https://github.com/JPZV/Blue-N64-Control-ESP32) (I'll add a pull request with the re-map after looking about this issue) and I found out something concerning about the n64 default mapping.

According to [n64.c Lines 76-77](https://github.com/darthcloud/BlueRetro/blob/master/main/adapter/wired/n64.c#L76-L77), the L/R Buttons are mapped to PAD_LS/PAD_RS respectively, while the Z Trigger is mapped to both PAD_LM and PAD_RM.

But, according to [gc.c Lines 65-67](https://github.com/darthcloud/BlueRetro/blob/master/main/adapter/wired/gc.c#L65-L66), the L/R buttons are mapped to PAD_LT/PAD_RT respectively, while the Z Trigger is mapped to both PAD_LS and PAD_RS.

This makes and issue when using an original N64 controller on both N64 and GameCube consoles with BlueRetro, as the Z Trigger (on the controller) will act as L+R (on the console), while the L and R buttons will do nothing.

My suggestion would be to change n64.c mapping to match to the gc.c one (only for the triggers). i.e.:

Before:

```cpp
BIT(N64_Z), BIT(N64_L), 0, 0, //Line 76
BIT(N64_Z), BIT(N64_R), 0, 0, //Line 77
```

After:

```cpp
0, BIT(N64_Z), BIT(N64_L), 0, //Line 76
0, BIT(N64_Z), BIT(N64_R), 0, //Line 77
```

But I wonder if that could break something with others controllers/custom-made mappings

darthcloud, 2023-01-02:
Default mapping are fine as they are.

n64.c is done so the mapping is good by default with any controller.

using n64 controller on anything but a n64 is an edge case not worth messing everything else for.

If someone want that they have the mapping config to fix it.

darthcloud, 2023-01-02:
also note that the adapter/wired/ source foles only affects their corresponding console. n64.c as zero impact on GameCube stuff.

Switch N64 BT controller is handled by adapter/wireless/sw.c

closed 2023-01-02 by darthcloud

JPZV, 2023-01-02:
That's the problem, I'm not talking about the N64 BT Controller, but a self-made one acted as a Generic controller. Right now I have this on mapping_quirks.c

```cpp
static void n64_bluen64(struct raw_src_mapping *map) {
    map->mask[0] = 0x771F0FFF; //Using GC mask. Maybe it should be ajusted

    map->btns_mask[PAD_RB_LEFT] = map->btns_mask[PAD_RB_RIGHT];
    map->btns_mask[PAD_RB_UP] = 0;
    map->btns_mask[PAD_RB_RIGHT] = 0;

    map->btns_mask[PAD_LT] = map->btns_mask[PAD_LS];
    map->btns_mask[PAD_RT] = map->btns_mask[PAD_RS];
    map->btns_mask[PAD_LS] = 0;
    map->btns_mask[PAD_RS] = BIT(8);

    map->btns_mask[PAD_MM] = BIT(5);
}
```

This will work perfectly on any GameCube console/adapter, but it will not work on a N64 **console** because L/R (physical buttons) does nothing, while Z Trigger act as L+R (i.e. every time I press Z, the L and R buttons will be pressed on the game).

So, what is your recommendation for that? Making the user choose/re-map everything? Because that actually is not a good solution.

darthcloud, 2023-01-02:
Setup your quirk so that L and R goes to LS & RS, and that z goes to RM.

You really don't need to touch stuff outside the quirk.

darthcloud, 2023-01-02:
RT LT is a special things for analog trigger with a end digital buttons. Dont use that.