darthcloud/BlueRetro #527: [GameCube] Action Replay not booting after installing BlueRetro Internal.

SuperSpunz, 2023-01-02:
Hello,
I installed a GameCube BlueRetro Internal from Laser Bear earlier this month, and just noticed my Action Replay won't boot anymore. I use the Action Replay with SD Media Launcher to load Swiss. I've tested on several GameCubes, and switched back to stock controller port board on one of them, so I'm pretty sure the BlueRetro is the conflict, but I'm planning on doing more testing when I can. 
Any ideas why this would conflict? 

In the meantime, I have switched to Wind Waker Hack for loading Swiss, but I would like to use the Action Replay if possible.

darthcloud, 2023-01-02:
My action replay work fine with BlueRetro. No reason for conflict.

lukas227, 2023-01-19:
@SuperSpunz I have exacty the same issue after installing the internal mod. However, I'm able to still boot the Action Replay disk by hitting the reset button once when the GameCube seems to "hang" after booting Action Replay.

I'd be curious to know if you are also able to boot successfully into the Action Replay by hitting reset.

I had two of the boards available (was experimenting with a friend) and Action Replay was "hanging" with both of the boards. Switching back to the original board resolved it again (Action Replay boots without needing a reset).

SuperSpunz, 2023-01-21:
@lukas227, After a bit more testing, this is exactly what I saw as well. I also noticed, powering back on after using the remote power down feature, it doesn't seem to hang when loading the Action Replay. It only hangs when using the GameCube's power button.

I also had 2 BlueRetro boards and 2 Action Replay discs to test, and saw the same thing on all of them.

I'm also curious, @darthcloud what region Action Replay did you use for your testing? I'm using NTSC-U, and I wonder if other regions don't hang.

darthcloud, 2023-01-21:
I think there is various versions of those disc maybe? Not sure how to check but I use mine mainly for testing the GC keyboard emulation.

Mine NTSC-U aswell.

lukas227, 2023-01-21:
I was testing on a PAL cube.

Just checked the redump.org database and at a quick glance it seems to list two different versions for the Action Replay in the US (1.06 and 1.20) but just one version for Europe (though there is no guarantee redump.org is complete).

Once I get to it I want to make a dump of my action replay and see if I can reproduce this on my own cube that has a GC Loader in it.

geekinballs706, 2023-02-03:
I’m having this exact issue after installing my GC internal blue retro except mine is with 007: Agent Under Fire. I use the save game exploit with this game and it’s the only game out of about 6 different GC discs I tried that needed me to reset after cold boot.

darthcloud, 2023-02-16:
I think v1.8.3 should fix this, download on https://darthcloud.itch.io/blueretro

lukas227, 2023-03-31:
Thanks a lot, I tested this today and can confirm this is fixed on my end! Sorry for the long delay, had to send in my board for repair and wasn't able to test for a long time.

closed 2023-03-31 by darthcloud