darthcloud/BlueRetro #533: [GameCube] Analog sticks and triggers calibration issues

Havelock-Vetinari, 2023-01-08:
Hi!

I'm using  BlueRetro board from Laser Bear Industries: https://www.laserbear.net/products/gamecube-blue-retro-internal-adapter
With version 1.8.1.

I've found that analog sticks and triggers are not calibrated well. I've been testing it with PS4 DualShock Original Gamepads (V2) and 8Bitdo GBross with aftermarket GameCube controllers.

It seems, that Blueretro board is reporting lower values for analog sticks and triggers:

Triggers (both left and right on stock BlueRetro setting report maximum of 208. Directly connected aftermarket gamepad reports 255 as max value.
When I've adjusted Max settings in Advanced Config to 125% and later 135%, the max reported value for triggers was 223. More over with this adjustment, after reaching max value 223, there is small dead section for trigger which  reports 0 as value, and them 223 once again when further pushed.
Stock configuration for triggers is missing digital mapping for L and R.
Tested on two PS4 DualShocks and on two 8Bitdo GBross adapters with aftermarket gamepads
All of above issues are not applicable when gamepad is directly connected to GameCube

Left and right analog sticks are also not calibrated. Directly connected gamepad reports up to +/-116 for both sticks. When connected wirelessly (both PS4 DualShock and 8Bitdo GBross) reports maximum of:  
- 100 lor left stick
- 90 for right stick

This can be mitigated by adjusting Max value by 115% and 125% respectively 

All test were performed on DOL-001 EU, with Swiss Swiss v0.6r1400 and with https://gbatemp.net/threads/gc-controller-test.365086/

Havelock-Vetinari, 2023-01-15:
Tested with latest 1.8.2 firmware. Still getting issues with analog triggers and sticks.

AwkwardHugger, 2023-02-07:
I am also experiencing calibration issues with trigger calibration. In Metroid Prime 2 I cannot lock-on because the triggers can be fully pressed. I am experiencing this with multiple controllers. Results on a PS4, PS5 and Xbox Series X controller.

GlassWolff, 2023-02-11:
*Bump*

Hoping for someone to notice and suggest something on this. I have the same issue on Switch Pro controller on GC. Triggers aren't working fully.

Havelock-Vetinari, 2023-08-20:
Issue still exists in v1.8.4 hw2 gamecube

darthcloud, 2025-04-22:
I feel theses problem got resolved in latest beta in the context of https://github.com/darthcloud/BlueRetro/issues/1015

The important thing to keep in mind with the triggers & joysticks are:

You need to use the merged preset to get the end click of the trigger:
[Wiki: BlueRetro BLE Web Config User Manual (33 example preset load)](https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#33---example-preset-load)

Also another important note for Xbox controller is that you need to press a digital button like A first a couple time before touching the triggers and joysticks to allow blueretro to calibrate them properly. This is noted here:
[Wiki: Controller pairing guide (4 xbox pairing guide)](https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#4---xbox-pairing-guide)

Finaly BlueRetro emulate an official Gamecube controller, those controller joystick do no not use the full 0-255 range possible, it's less than that. Once signed left stick is got a range of -100 to 100. Only 3rd party controller would give -128 to 127.