darthcloud/BlueRetro #536: 8bitDo Saturn DIY controller captures

svirant, 2023-01-12:
The "8bitDo Saturn DIY" controller has 4 different controllers modes. I have captured
each one in their respective modes.  Below is the order of buttons pressed:

UP
DOWN
LEFT
RIGHT
A
B
C
X
Y
Z
Start
L
R

[8bitDo-Saturn-DIY.zip](https://github.com/darthcloud/BlueRetro/files/10402901/8bitDo-Saturn-DIY.zip)

darthcloud, 2023-01-12:
Awesome thanks!

darthcloud, 2023-01-15:
v1.8.2 now support this controller natively. https://github.com/darthcloud/BlueRetro/releases/tag/v1.8.2

The controller need to be configured to expose the d-pad as a d-pad:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#73---d-pad-as-joystick-or-d-pad-configuration

closed 2023-01-15 by darthcloud