darthcloud/BlueRetro #538: Não consigo conectar 2 controles bluetooth ao mesmo tempo, para poder jogar duas pessoas.

**I can't connect 2 bluetooth controllers at the same time, so I can play two people.**

Alefox007, 2023-01-12:
Não consigo conectar 2 controles bluetooth ao mesmo tempo, os dois controles são da marca "ipega" e de modelos diferentes.O primeiro é "Ipega 9025"' mais antigo, e o outro é o "Ipega 9076", os dois controles funcionam perfeitamente porém só consigo conectar apenas um deles. Resumindo: Os dois controles conectam perfeitamente, mas não consigo parear os juntos para jogar jogos multiplayers.

I can't connect 2 bluetooth controllers at the same time, the two controllers are from the "ipega" brand and different models. The first is the older "Ipega 9025", the other is the "Ipega 9076", both controllers work perfectly but I can only connect only one of them. Bottom line: The two controllers connect perfectly, but I can't pair them together to play multiplayer games.

hardlevel, 2023-01-16:
você so pode conectar dois ou mais em blueretro que tem duas ou mais conexões de cabo, aquele simples do aliexpress pra ps1 e ps2 que serve so em um slot não da pra conectar dois

Alefox007, 2023-01-16:
> você so pode conectar dois ou mais em blueretro que tem duas ou mais conexões de cabo, aquele simples do aliexpress pra ps1 e ps2 que serve so em um slot não da pra conectar dois

O meu é aquele núcleo que vem com o cabo a sua escolha, no meu caso escolhi de PS2 que conectam nos dois slots, o jogo até reconhece que tem controle porém não consigo parear os dois simultaneamente...

hardlevel, 2023-01-16:
veja nas configurações na interface web la conectando pelo celular, se a
opção do multitap tá ativa, tem que estar em none, o multitap não funciona
no ps2

Alefox007, 2023-01-16:
> veja nas configurações na interface web la conectando pelo celular, se a
> opção do multitap tá ativa, tem que estar em none, o multitap não funciona
> no ps2

**Já está a assim, sem sucesso ainda...**

[image]

Alefox007, 2023-01-16:
Já tentei colocar na configuração global o PS2 mais não vai também

hardlevel, 2023-01-16:
Vc já tentou configurar os dois outputs nessa tela aí da imagem? Configura
o 1 e o 2

Alefox007, 2023-01-16:
Hmmm, vou testar pra ver...

Alefox007, 2023-01-16:
É também não adiantou