darthcloud/BlueRetro #545: [Wired] Add Wii extension support for encryption mode

darthcloud, 2023-01-15:
Would need to integrate Marcan code https://github.com/dolphin-emu/dolphin/blob/master/Source/Core/Core/HW/WiimoteEmu/Encryption.cpp

But it's GPL and not willing to change the project license.

Could ask Marcan if possible to dual license the code.

Invictaz, 2023-01-15:
Implementation is here
http://web.archive.org/web/20111210161944/http://pastie.org/private/6vajs4t1glpd2fome7x1g

Writeup here

darthcloud, 2023-01-16:
Marcan is ok with licensing his code under MIT, so that might come in future if the impact on resources usage is acceptable.

[image]

commit 5e9759e7 (2023-01-17): [WIRED] Add support for Wii extension encryption

Closes #545
closed 2023-01-17 by darthcloud