darthcloud/BlueRetro #547: Exlene rumble not working in Xinput mode

darthcloud, 2023-01-15:
To be investigated why rumble doesn't work

aleomark, 2024-07-04:
Im on firm 24.04_hw2, blueretro internal GC, Exlene rumble not working, when imput in home + Y, and home + A

darthcloud, 2025-01-24:
Fixes in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)

closed 2025-01-24 by darthcloud

icd2k3, 2025-01-26:
Sorry to dig this back up - I updated to this firmware, set presets for Exlene output in the blueretro UI, but I can't seem to get the rumble to work (I probably just have an incorrect setting or something)

darthcloud, 2025-01-26:
Use switch mode:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#12---exlene-gamecube-pairing-guide

icd2k3, 2025-01-26:
Thanks for the reply @darthcloud! - I did a factory reset and re-paired the Exlene controller - I believe it's in Switch mode now because I have to press (-) to get the correct rotated name based mapping like the docs mention (and it rumbles when I press it)... but oddly still no rumble in-game 🤷‍♂ tested in Meele and Sonic Riders

darthcloud, 2025-01-26:
you need to enable rumble in the config
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#26---example-enable-rumble

icd2k3, 2025-01-26:
Ah right, thanks (sorry been a while since I've changed these settings lol) - I ended up switching back to 24.10 because I forgot about the multiple Switch controller latency issue. Is there a way to stay on the current firmware, but connect the Exlene in X-input mode like 24.10? If not, I'll just stay on that one for now NBD

Thanks for your work on this project - I have it in all my consoles and has worked flawlessly for years!

darthcloud, 2025-01-26:
xinput will work on v25.01 juat like before

darthcloud, 2025-01-26:
Some additional info:
The xinput (aka ios) mode of the controller do not expose a rumble report.
So it's impossible to have rumble in that mode.

Only Switch mode can have rumble.

icd2k3, 2025-01-27:
Makes sense, still having trouble with 25.01 though

- 4 Exlene GC controllers
- Factory reset BR
- update to 25.01
- in presets config, change outputs 2, 3, and 4 to preset "Exlene GameCube"
- Re-pair all controllers one by one
- Try to connect all controllers, and the 4th connection causes a full reset of BlueRetro

icd2k3, 2025-01-27:
UPDATE: ok I think I finally got this right haha, paired controllers 2-3 via Y + home and now it seems good (player 1 rumbles, no rumble on the other 3, no latency)