darthcloud/BlueRetro #548: RF Brawler64 NSO switch mode rumble don't stop sometimes

darthcloud, 2023-01-15:
To be investigated.

Workaround 1 press Home couple time to turn rumble off and on again.
Workaround 2 Leave rumble off

skite2001, 2023-01-16:
ah...you already found this bug. i will delete my issue and merge it here:

First of all, thx @darthcloud for your amazing firmware update! controls are working almost nicely now
(dont know why the cbuttons dont work correct on zero hour..but thats another story)

Did some rumble testing (Switch-Mode) with Starfox, Waverace and Duke Nukem:ZeroHour.
While in overall the rumble works, there seems some loop issues, where the rumble doesnt stop at all. Sometimes i had to use the rumble toggle switch, to turn it off and on again.
For example with starfox...
Go into training mode and hold your laser button to charge it up.
After shooting this big green spark and impact, rumble seems to be endless some times.
Same issue happens to waverave, but its more variable and i cant say exactly when it happens.
On Zero Hour the rumble doesnt seem to work at all. Tried to start it normal, nothing.
Tried to press the rumble toggle button at the "please insert rumblepak" screen. nothing.
blueretro is set to both (Mempack+Rumblepack)
Using an ED64+ Everdrive

darthcloud, 2023-01-16:
@skite2001 (Mempack+Rumblepack) is not a valid setting. You need to be set to either rumble or memory in the config or use the Home button to toggle between the 2

skite2001, 2023-01-16:
> @skite2001 (Mempack+Rumblepack) is not a valid setting. You need to be set to either rumble or memory in the config or use the Home button to toggle between the 2

ah, ok...i thought the "Mode: Gamepad - Accessories: Both" work that way

darthcloud, 2023-01-16:
On a real controller you can't fit both a rumble and mem pak in the same slot.
That setting is intended for Dreamcast only

See doc https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-System-Specific-User-Manual#accessories-14

patrickmollohan, 2023-03-15:
Can confirm this happens on the GameCube internal adapter from Laser Bear Industries with an Exlene GameCube controller in Switch mode (accessory set to "Rumble"). Certain actions in games will activate the rumble for about 20ish seconds before it eventually stops, while other actions seem to be just fine. Using firmware 1.8.3.

darthcloud, 2023-12-09:
Fix is now in v1.9.1

closed 2023-12-09 by darthcloud