darthcloud/BlueRetro #553: PowerA Gamecube Bluetooth controller add digital L and R keys

scp-electrocution, 2023-01-16:
Hi, I'm using PowerA Gamecube Bluetooth controller (for Switch). Unfortunately it is not possible to map the digital L and digital r buttons even if the controller physically has them (analog l and analog r work correctly). Unfortunately the controller is not compatible with windows and I can't use the tools you provided to map the keys, would it be possible to implement support for this controller in a future blueretro release?
It would be great for games that require those buttons like Super Mario Sunshine for example.
Controller link: https://www.powera.com/p/nintendo/nintendo-switch/controllers/wireless/wireless-gamecube-style-controller-for-nintendo-switch/
Thank you

darthcloud, 2023-01-16:
Load the gamecube merge trigger preset:
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#33---example-preset-load

rafaeoz, 2023-01-17:
> Load the gamecube merge trigger preset:
> https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#33---example-preset-load

I have this version ![IMG_20230116_235428.jpg]
(https://user-images.githubusercontent.com/114241445/212799619-b038365e-9d36-4efe-9ef1-4988237e8333.jpg)

Lr digital is not mapped to ps4 and ps3 . Should i use that tutorial too?

And i have trouble to sync with Bluetooth. Just happened a few times.

darthcloud, 2023-01-17:
Yes you need that preset for all controller lacking a end digital buttons which is pretty much all.

Exception is Wii classic OG and GameCube wires via Gbros adapter

closed 2023-01-17 by darthcloud