darthcloud/BlueRetro #555: Add support for Ouya controller.

keiem, 2023-01-17:
Is it possible to add support for the ouya controller?  I have a few of these controllers and would like to use them with my gamecube.

darthcloud, 2023-01-17:
Get me a trace of it see:
https://github.com/darthcloud/BlueRetro/wiki/Bluetooth-HCI-trace-with-Win10

keiem, 2023-01-17:
I just found that page myself, but you were quicker.  I will dig up somewhere a Win10 pc  and get the HCI-trace of the ouya controller.
Just for you info: I can connect the ouya controller to blueretro, but not all buttons work.

wfowler1, 2023-01-17:
I can also get a trace of this, I was thinking of making a bug report for this controller but I was too lazy and/or forgot. I've used Wireshark before to get packet grabs to help the OpenRGB project so I can probably figure it out for this as well.

For the record, I wasn't able to connect my Ouya controller to the LaserBear GameCube thingy at all, it simply wouldn't sync.

keiem, 2023-01-17:
Herewith the trace and the numbering.  I hope I did everything correct.
[ouya controller.pdf](https://github.com/darthcloud/BlueRetro/files/10438800/ouya.controller.pdf)
[ouya_trace.pcapng.zip](https://github.com/darthcloud/BlueRetro/files/10438804/ouya_trace.pcapng.zip)

darthcloud, 2023-01-17:
Trace is good thanks!

darthcloud, 2023-10-07:
Sorry for long delay, I added support base on the trace.

See beta here:
https://github.com/darthcloud/BlueRetro/actions/runs/6443343629

I couldn't see L3 and R3 press in the trace. Maybe the joystick buttons aren't wired to anything?

Some bits were not used so I defined them as L3 and R3 just in case.

keiem, 2023-10-08:
No problem and thanks.

I think there wa a error in the trace, as I had to clean the contacts of my ouya controller a few days after I made the trace, because sometimes my right shoulder button didn't work during Mario Kart.  So probably it was also the case when I did the trace.  I can redo the trace.

darthcloud, 2023-10-15:
Fixed in v1.9 https://github.com/darthcloud/BlueRetro/releases/tag/v1.9

closed 2023-10-15 by darthcloud

keiem, 2023-12-15:
@darthcloud 

Jacques, can you check my new traces.  The L2 and R2 buttons don't seem to work on the GameCube with version 1.9.  Maybe you find in this one L3 and R3.
I made2 traces, just to be sure all buttons are detected.
[ouya trace.zip](https://github.com/darthcloud/BlueRetro/files/13686228/ouya.trace.zip)