darthcloud/BlueRetro #558: [Gamecube] HW2 Shutdown Button Combination Collision

messem10, 2023-01-18:
For HW2 setups, the power shutdown button combination collides with that of *Super Smash Bros Melee*’s back to menu command. (L+R+A+Start)

Would suggest changing it to something else to avoid inadvertent shutdowns, especially in a popular game.

darthcloud, 2023-01-18:
v1.9 will make combo function customizable.

As a workaround you could map one of the unused button to make the combo via the advance config mapping.

Example with playstation label for SRC and gamecube label for DST:
SRC DST
L1 L digital
L1 R digital
L1 Start
L1 A

Click the "+' button in web page to add extra mapping. Leave all other columns to default value.

with that mapping pressing L1 will result in combo L+R+Start+A

darthcloud, 2023-10-15:
You can now modify the macro buttons in v1.9

See https://github.com/darthcloud/BlueRetro/releases/tag/v1.9

closed 2023-10-15 by darthcloud