darthcloud/BlueRetro #566: The X-Files PS1 with BlueRetro & 8BitDo Pro 2 no IGR

L10N37, 2023-01-19:
Kills my reset combos for IGR
Only game I know of.

If you know how to enable rumble (in-game) also, let me know :)

I get a rumble to let me know the controllers synced, but nothing in game. I have the controller set to "X" on the rear switch.

L10N37, 2023-01-19:
Ok, I read up and saw the OTA updates (only one of my BlueRetro dongles has USB) and although I lost my BT dongle for the desktop and i CBF setting up my laptop that's being unused since moving a few months back - I used my phone. It only took a couple minutes to update using Chrome on android, which was good. One dongle was even faster!

So I'm on 1.8.2 HW1 now and used the BlueRetro_playstation.bin (they both came with 1.7.1 psx_external)
Rumble enabled in the settings now-  though still no rumble, which is a little disappointing.. 

As for the IGR, the X-files seems to enable analogue mode on the controllers once it gets past the intro video/FMV. 
Prior to this you can reset with the combos.

Off memory the IGR is simple and obviously with a product/ project like this, you'd know all about it

it's a few constant bytes and the 2 variables for button input

1: FFh   2: 41h (digital)/73h (analogue) 3: 5Ah  4: Inputs  5 Inputs 

So

Long story short, i disable analogue and the combos work.
I tried in another game, analogue mode broken on that too(weird, i tested both modes before finalising the tiny bit of code and re-assembling the console). 

I will debug it eventually... but i guess this is case closed. Really confused about this one!

If you get rumble working on 8BitDo Pro 2 that would be awesome :)

darthcloud, 2023-01-19:
Ok and I guess the IGR do work with a real DS1 in analog mode?

L10N37, 2023-01-19:
No, but AFAIK analogue wasn't automatically switched on using a genuine controller on that game.

I'll have a proper check tomorrow.

It was definitely working prior to reassembly so a little confused here. Was certain both modes were tested after assembly too...oh well

closed 2024-04-15 by darthcloud